OtherCrashOverride/go-play #1: Setup documentation?

gbanfalvikl, 2018-06-26:
Would it be possible to add some basic documentation about how we can build this ourselves?

OtherCrashOverride, 2018-06-27:
There is discussion about building here:
https://forum.odroid.com/viewtopic.php?f=159&t=31303#p226519

Feel free to note anything that needs further clarification.  The short explanation is that you set up an esp-idf build environment and then change into the directory of the project to build (nesemu-go, gnuboy-go, smsplus-go).  Each project is built the same as for any other esp-idf project (make).

rickyzhang82, 2018-06-30:
@OtherCrashOverride 
Can you open a wiki here? I will copy your notes to move it to your repo.

rickyzhang82, 2018-07-01:
@OtherCrashOverride Please allow others to create a wiki page. I can't do the leg work.

OtherCrashOverride, 2018-07-01:
I changed the wiki setting.

closed 2018-07-09 by OtherCrashOverride