OtherCrashOverride/go-play #6: [NES] Super Mario Bros. 3 not loading

kamotswind, 2018-06-30:
So far all games I have tried have worked very well, great job! The only game that I am having issues with so far is Super Mario Bros. 3 which won't load.

When the game loads it just freezes at the "red curtain" screen. I can see parts of the display twitching a little. I even tried out the pre-release firmware (20180629-psram) and it actually got worse. At first it seemed to load ok but as soon as I tried to play level 1 it froze completely including the audio track. I had to turn off the power since none of the buttons would respond.

I rolled back the firmware to the current stable release (20180627) and now when I load the game it just shows sprite garbage on the screen and hangs.

If you need more information please let me know.

OtherCrashOverride, 2018-06-30:
https://forum.odroid.com/viewtopic.php?f=159&t=31379

kamotswind, 2018-06-30:
I can confirm the instructions you provided on the forum works to resolve the issue. Thank you!

kamotswind, 2018-07-04:
@OtherCrashOverride Do you hang out on Freenode or some other place? Would like to collaborate if you are interested.

OtherCrashOverride, 2018-07-04:
I joined the #odroid channel on Freenode.

OtherCrashOverride, 2018-07-05:
Resolved by 197f806ca0efe544cb3231d2667717d3ea1681d8

closed 2018-07-05 by OtherCrashOverride