OtherCrashOverride/go-play #8: [Proposal] Artificial Red "power" light during GB/GBC emulation

Ongnissim, 2018-07-01:
Hey all, would it be possible to add a cluster of red pixels to the left hand side of the emulated Gameboy screen to simulate the Power LED on actual gameboy hardware?

The Pocket Sprite does something similar in it's implementation of Gnuboy, and I think it's a neat touch that adds asimple nostalgic flair.

Thanks!

OtherCrashOverride, 2018-07-01:
My concern with this feature would be LCD "burn in".

OtherCrashOverride, 2018-07-03:
The easiest place to do it would be **before** the following line:
https://github.com/OtherCrashOverride/go-play/blob/eb05c0a1faf67728bb65166953d80ffc470b6d80/odroid-go-common/components/odroid/odroid_display.c#L555

You would create a bitmap (RGB565) that has the shape and color of the power indicator and draw it with:
https://github.com/OtherCrashOverride/go-play/blob/eb05c0a1faf67728bb65166953d80ffc470b6d80/odroid-go-common/components/odroid/odroid_display.h#L20

The "LE" means "little endian", if your bitmap is "big endian" use:
https://github.com/OtherCrashOverride/go-play/blob/eb05c0a1faf67728bb65166953d80ffc470b6d80/odroid-go-common/components/odroid/odroid_display.h#L18

closed 2018-07-09 by OtherCrashOverride