OtherCrashOverride/go-play #9: Investigate I2C RTC support for GB/GBC

OtherCrashOverride, 2018-07-01:
The ODROID-GO has an expansion port on the top.  It should be possible to attach an RTC to this for games that make use of the date/time.

kamotswind, 2018-07-04:
Is the code for the base OS on GitHub? I may be able to help with this since I have worked with I2C RTCs before and I have several. I haven't worked on an ESP32 platform before but it doesn't look much different.

OtherCrashOverride, 2018-07-04:
The base OS code fork is here:
https://github.com/OtherCrashOverride/esp-idf

It should not require any OS modifications to implement.  Using the I2C APIs, the RTC can be read and the date set using an API call.

rickyzhang82, 2018-07-04:
How about enable WIFI and use NTP to get a time stamp?

OtherCrashOverride, 2018-07-04:
Although possible, the use of wifi would introduce a large delay to game startup during the associate and retrieval operation.  It would also require the introduction of UI to display APs and allow entering a password to connect.