OtherCrashOverride/go-play #10: Performance issues in 3d segments of Phantasy Star

BakaOsaka, 2018-07-02:
I am currently running Phantasy Star (English) on version 20180605 of the firmware using the Sega Master System emulator.
The game appears to run fine for the most part, however when you enter the pseudo-3d maze segments of Phantasy Star the FPS will drop until the scene has rendered. Following that, it will stutter every time you attempt to navigate the maze. Standing still (ie, not pressing any inputs) will run correctly and will not lag at all.

OtherCrashOverride, 2018-07-02:
I need a copy of the .sav file prior to where the issue occurs so that I can easily and repeatedly reproduce the issue for testing.  The .sav file is created when you press "Menu".  Note that you must load a different ROM after saving to move the save file from internal flash to SD card.  Once on the SD card, it should be located in "odroid/data/sms" with the name of the ROM file ending with .sav.

kamotswind, 2018-07-04:
I can also reproduce this fairly easily. In the beginning area there are these underground tunnels or something that you can enter. I can try to take some screen shots with instructions if that doesn't help you reproduce it.

BakaOsaka, 2018-07-04:
Please find attached the relevant save file.
Simply moving forwards or backwards at this current point in the game will trigger the issue - FPS will drop and the audio will stutter.

[Phantasy Star (USA, Europe) (v1.3).sms.sav.zip](https://github.com/OtherCrashOverride/go-play/files/2162430/Phantasy.Star.USA.Europe.v1.3.sms.sav.zip)

OtherCrashOverride, 2018-07-05:
Resolved by 73337cb7803f38fe15eacbe09a2816ed9f62009c

closed 2018-07-05 by OtherCrashOverride