OtherCrashOverride/go-play #12: Game Boy  color palette

Crushless, 2018-07-02:
It would be nice, to have an option to change the color palette in Game Boy emulation. Just like the Super Game Boy for the SNES did. If this would be too complicated, you could add an option to cycle through a set of predefined color palettes. At least please add a color palette with a light green background, like the original Game Boy.

OtherCrashOverride, 2018-07-03:
Someone will need to provide the RGB color values for the palettes.

Crushless, 2018-07-03:
Unfortunately, I don't have the exact RGB codes ready, but I found some website that lists all the predefined palettes:
https://www.mariowiki.com/Super_Game_Boy#Color_palettes
And this website lists all GB Color palettes:
https://en.wikipedia.org/wiki/List_of_video_game_console_palettes#Game_Boy_Color

Crushless, 2018-07-03:
Referece and background info about the Super Game Boy palettes, with RGB codes:
https://gamefaqs.gamespot.com/snes/588731-super-game-boy/faqs/58647

kamotswind, 2018-07-04:
OtherCrashOverride provided this link to the current palette:
https://github.com/OtherCrashOverride/go-play/blob/master/gnuboy-go/components/gnuboy/lcd.c#L52-L54