OtherCrashOverride/go-play #13: [NES] Display aligment

puggile, 2018-07-04:
Hi,
I've noticed that nes emulator has not perfect alignment in the display.
You can see easily in game like Castlevania where on the top-left on the screen there are player stats truncated. Like the last pixel line is cut.
If i play in scaling mode the game is perfect.

firmware version: 20180702-psram

OtherCrashOverride, 2018-07-04:
When full screen, 4 pixels are removed from each side of the display:
https://github.com/OtherCrashOverride/go-play/blob/eb05c0a1faf67728bb65166953d80ffc470b6d80/odroid-go-common/components/odroid/odroid_display.c#L1047

This was done for performance reasons as scaling is done on the CPU.

puggile, 2018-07-05:
thanks for explanation!

closed 2018-07-05 by puggile