OtherCrashOverride/go-play #14: DAC query/possible sound bug

DavidKnight247, 2018-07-04:
I am reviewing the code and don't understand this part.

https://github.com/OtherCrashOverride/go-play/blob/eb05c0a1faf67728bb65166953d80ffc470b6d80/odroid-go-common/components/odroid/odroid_audio.c#L57

The ESP32 documentation states the DAC is 8bit yet you define a 16bit here. sound is clearly working but I can hear lots of crackle.

Is this because the DAC is only reading the first byte?

OtherCrashOverride, 2018-07-04:
That is a nuance of the esp-idf SDK.  The sound first goes to the I2S driver that does not accept 8bit (at the time of development).  The hardware then routes the top half (8bit) to the built in DACs (also 8bit) instead of an I2S peripheral.

OtherCrashOverride, 2018-07-04:
I should further note that since the volume is digital, a reduction in volume is also a reduction in bit depth.  Sound should not crackle when the volume is 100%.  If the sound is broken/choppy at 100%, then its due to emulation speed.

DavidKnight247, 2018-07-04:
Thankyou, this makes sense now! I was struggling to understand the reason for the crackling sound.
FYI i2s_write_bytes is deprecated in favour of i2s_write.

DavidKnight247, 2018-07-05:
Hi again,
I've made a few changes to audio.c with comments and a few minor alterations. I tried to generate a pull request but I haven't created a branch as I haven't used guthub for a few years :-(

Please feel free to merge any code you are happy with and ignore the code you are not happy with!

Feedback much appreciated but not expected, I'm just happy to be coding again :-)

https://github.com/DavidKnight247/Odroid-Go/blob/master/odroid-go-common/components/odroid/odroid_audio.c

OtherCrashOverride, 2018-07-06:
The change appears to swap the order of operation for the divide in half and multiply by volume.  This should result in one extra bit of precision, but the entire 8 lower bits are discarded.  Therefore, I do not think it would produce any actual change in the values.

DavidKnight247, 2018-07-06:
Thankyou, one final suggestion.

Changing the volumeLevels array values to [0, 125, 250, 500, 1000] produces better sound as each represents a clean reduction of 1 bit in the audio signal. Additionally we can then use bitshifting instead of floating point maths which would be computationally far more efficient.

I have tested volume levels of 125 and 250 and both sound cleaner than 150. If you are wanting to stick with 4 sound values I'd suggest 250 but it would be better IMO to offer 5 sound settings.

Please let me know if you feel this would be worthwhile and I will prepare a patch.

OtherCrashOverride, 2018-07-06:
The ESP32 has a math co-processor.  The use of "single" precision floating point should be a single machine instruction.

Fork the repo.  Create a branch with your suggested changes.  Then submit a pull request.

I will test the changes and collect feedback.

DavidKnight247, 2018-07-06:
https://github.com/OtherCrashOverride/go-play/pull/20

DavidKnight247, 2018-07-06:
A warning: because I am changing the number of volume levels this will cause instability if volume is set to 4 when returning to springboard. This may cause problems for testers.

OtherCrashOverride, 2018-07-07:
It would be nice if someone could measure the decibel levels output on hardware for each level.  Im not convinced there is enough difference to warrant an extra sound level.

OtherCrashOverride, 2018-07-07:
Something else to consider is that in the future I want to support an external DAC connected to the expansion port.  In this situation, there is no mono down mix. So, the code path is unique to internal DAC use.  However, I would like the volume settings to also be applicable to an external DAC.  I do not think bit shifting would be appropriate for it since it discards so much dynamic range of the 16bit samples an external DAC would use.

I don't see any issue with accepting a patch that swaps the operation order to gain an extra bit of precision in the calculation.  However, I am inclined to keep the volume as a lookup-table and floating point operation.  This preserves more fidelity when an external DAC is implemented.  It also allows the flexibility of "profiles" in the future instead of hard coded volume levels.

DavidKnight247, 2018-07-07:
So keep float calc for external DAC. 
For internal DAC this is more efficient in processor and memory usage. 
Then perhaps if we confine bitshift to BUILTIN_DAC_ENABLED code float can still be used later for external DAC?
It would be a good idea to quantify efficiency gains to check if they are significant. There’s little point increasing code complexity if the gain isn’t worthwhile.

OtherCrashOverride, 2018-07-07:
Since the code is not currently a bottleneck, there is little to gain from altering it.

Sound improvements are more likely to be gained by examining the hardware:
https://github.com/hardkernel/ODROID-GO/blob/master/Documents/ODROID-GO_REV0.1_20180518.pdf

The first page (top/left) shows the audio amplifier: PAM8304.  Notice that both lines of its differential input are connected to the ESP32.  Currently, IO25 provides a constant reference voltage while IO26 presents the sound samples.  By altering both IO25 and IO26 (dac0/dac1 in the code), more precise volume control can be achieved.  The catch is that IO25 also controls the amplifier suspend status.  So its range of values must be above a certain threshold for sound to be produced.  Once this threshold is know, a more advanced algorithm can be developed to offer improved sound quality.  This is the direction I would like improvements to the sound driver to take.

OtherCrashOverride, 2018-07-07:
I coded an experimental implementation of what was previously described:
https://github.com/OtherCrashOverride/go-play/commit/351a1d87dd86417810b92fec5e407a8b65a76bba

This effectively provides an additional bit of resolution.  It operates the DAC output as pure differential  rather than the bit centric method previously used.  Noteworthy is that the swing on IO25 is faster than required for the PAM to ender suspend mode allowing the full use of its range.

It should be possible to algebraically optimize the algorithm.  It was coded for clarity, not efficiency.

It also makes it worthwhile to review the audio level settings.
#3

kamotswind, 2018-07-07:
Just throwing in my 2c - it would be great if the lowest volume setting was about 50% lower then it is now (subjectively). In a mostly quiet room it seems too loud. To that end [0, 125, 500, 100] would be the replacement levels based on the old code. Sounds like you guys are doing some cool stuff to make it even better.

OtherCrashOverride, 2018-07-08:
After further testing, I think we need **both** changes to the sound volume and another sound level.

kamotswind, 2018-07-08:
If you point me in the right direction I'm happy to modify the sound volume level indicator on the home screen GUI along with related code to support the additional level.

OtherCrashOverride, 2018-07-08:
I will need to do the modification on the home screen.  The rest should be in the same place as the pull request mentioned earlier:
https://github.com/OtherCrashOverride/go-play/pull/20/commits

kamotswind, 2018-07-08:
I submitted pull request #25 - Hopefully it doesn't suck :)

All the changes were based on @DavidKnight247 's work, except for the typo corrections.

DavidKnight247, 2018-07-08:
"Noteworthy is that the swing on IO25 is faster than required for the PAM to ender suspend mode allowing the full use of its range."
I tested at the lowest volume setting (150). The audio in Adams Family on SMS sounds like it's cutting out in parts. This doesn't seem to affect 500 and 1000 values. 
Could this be the PAM cutting out - I presume there's no way to check this in software?

OtherCrashOverride, 2018-07-08:
The swing on IO25 only happens at the maximum extent (loud).  At reduced volume, operation should be the same with IO25 constant.

The choppyness at reduced volumes is due to the lack of bits as before.  This is why I stated elsewhere that volume changes need to be tested across all emulators and with popular games.

It may be possible to use the inverse of previous method to "shrink" the sound at reduced levels.  This will need to wait until after I make another test release.

kamotswind, 2018-07-08:
I have several games on each emulator and I'll be happy to test all volume levels on the next build. I really need to get my build environment going....

OtherCrashOverride, 2018-07-09:
Closed by 351a1d87dd86417810b92fec5e407a8b65a76bba

closed 2018-07-09 by OtherCrashOverride