OtherCrashOverride/go-play #15: Use and display sub folders

Voljega, 2018-07-05:
It might be good to be able subfolders in rom directories, it facilitates the navigation (I personnaly use it to sort games by genre)

For now they are not recognised at all, games in subfolders simply not appear, nor the subfolders themselves

OtherCrashOverride, 2018-07-06:
The issue with sub-folders is they add size to the path string.  This can cause an issue in the emulators where the RAM is very limited.  Therefore, I do not plan to support this feature for the technical reason stated.

Voljega, 2018-07-06:
oh ok, thank you for the response, I wasn't aware ram size and handling was that critical

Voljega, 2018-07-06:
What about limiting folder name to three chars ?

OtherCrashOverride, 2018-07-06:
It would lead to customer confusion if an arbitrary limit was used.  There is no indication of the limitation or explanation of why a folder does not show.

I should point out that its possible for anyone to write their own menu program with the features they desire.

closed 2018-07-09 by OtherCrashOverride