OtherCrashOverride/go-play #18: Display cover of the game

Voljega, 2018-07-05:
Although I have not too much of a good idea how to display cover of the game without masking too much of the game name on such a small screen, it would be nice to display cover for games

OtherCrashOverride, 2018-07-06:
This is a feature I wanted to implement.  There are just not enough resources available on the device to scrape for cover art.  A PC program that does the scraping and "baking" of the image (raw RGB565 of fixed size) will need to be developed to achieve this feature.

Voljega, 2018-07-06:
Yes, shouldn't be too complicated.
What size could the Odroid Go handle for the picture without hurting menu navigation performances ?

Also I suppose scrapping name for the games and displaying them is not possible once again regarding resources ?

OtherCrashOverride, 2018-07-06:
The image size will depend on how the feature is implemented.  Selecting files by cover art or "proper" name will likely not be possible due to resource constraints.  However, once a file is selected, the cover art could be displayed.

OtherCrashOverride, 2018-08-10:
Cover art is displayed in the 20180810 release.

closed 2018-08-10 by OtherCrashOverride