OtherCrashOverride/go-play #22: [GBC] Game & Watch Gallery 3 graphic problem

puggile, 2018-07-07:
I see a graphic problem in G&WG3 for GBC.
I provide you save file so you can easily reproduce the issue.

Please find attached file.
[Game & Watch Gallery 3 (USA, Europe).gbc.zip](https://github.com/OtherCrashOverride/go-play/files/2173192/Game.Watch.Gallery.3.USA.Europe.gbc.zip)

After you load save file, then you just see on the top/left the display the truck image is troncated.
fw version: 20180702-psram

For this kind of problem, do you prefer that I open a problem on Github or write the problem on public document on Odroid forum?

OtherCrashOverride, 2018-07-08:
I can reproduce the issue.  It looks like tile/sprite clipping.

Reference link:
https://www.youtube.com/watch?v=ug89k2FhiJU&feature=youtu.be&t=2475

Issue are best posted here since I can link changes to them.

OtherCrashOverride, 2018-07-08:
Resolved by 1f1edd355f48711c8efa12e72995d4e36b42d24d

closed 2018-07-08 by OtherCrashOverride