OtherCrashOverride/go-play #24: make fails

bornhaeusser, 2018-07-08:
Hi,
i want to make some new bin files out of this source.
I intalled the Toolchain etc. how in this tutorial on the wiki site.

$ make flash monitor
CC build/gnuboy/mem.o
C:/msys32/home/db/OdroidGo/Firmware/gnuboy-go/components/gnuboy/mem.c:17:38: fatal error: ../odroid/odroid_display.h: No such file or directory
compilation terminated.
make[1]: *** [/home/db/esp/esp-idf/make/component_wrapper.mk:286: mem.o] Fehler 1
make: *** [C:/msys32/home/db/esp/esp-idf/make/project.mk:468: component-gnuboy-build] Fehler 2

bornhaeusser, 2018-07-08:
ok thanks! - so is it possible to get a new prerelease? Thought i could this make myself :-)

OtherCrashOverride, 2018-07-08:
I was planning to make another release as soon as #14 wraps up.  It is now necessary to have the volume changes.

bornhaeusser, 2018-07-08:
Ok - Thank you!!

closed 2018-07-08 by bornhaeusser