OtherCrashOverride/go-play #26: Suggestion: Change volume control

DavidKnight247, 2018-07-08:
Currently volume level is determined by an array containing four values, 0, 150, 500 & 1000.
These values are multiplied by 0.001 to create a range between 0 and 1 with 0 representing mute and 1 maximum volume.

Human hearing is _non-linear_ i.e. logarithmic and the above values do not conform to this scale.

A reasonable approximation is to use 1000x^4 scale (see reference). Floor and ceiling values would depend on the DAC used. The 8bit internal DAC could perhaps use 0.6 <= x <= 1.0 whereas an external DAC with better fidelity could have a greater range.

Using this equation you can take the values of 0.6, 0.8 and 1.0 to produce equivalent (integer) volume levels of **130, 409 and 1000** for the 8bit DAC. If you prefer a floor value of 150 then you can use the values of 0.62, 0.81 and 1.00 for values of **148, 430 and 1000**.

If in the future fine-grain volume control is introduced the above equation would sound much better than if a linear scale were used. This could be simply implemented by continuously adjusting volume when the volume button is depressed.

I am happy to create this code of course :-)

**Reference:**
https://www.dr-lex.be/info-stuff/volumecontrols.html

OtherCrashOverride, 2018-07-08:
Ok, I am going to take the pull request increasing the volume level count (#25) and base the next test release on code at that point.  Once that patch is in, you can submit a pull request "tuning" the volume levels.

DavidKnight247, 2018-07-08:
Great!

runnerpack, 2018-07-09:
I had an idea, recently: while holding the sound button, up and down on the d-pad control the volume with finer granularity. It wouldn't have to replace the current "cycling" mode, but allows for finer adjustment when needed.

DavidKnight247, 2018-07-09:
I tested this whilst evaluating the code. I personally agree but like
changing the brightness it isn’t intuitive for the user. Perhaps a graphic
could be displayed to prompt the user?

kamotswind, 2018-07-09:
@DavidKnight247 You may want to try the latest release. The sound values you provided sound very good to me. IMHO having variable volume or messing with the volume levels again just doesn't seem worth it. I'd rather see a few more emulators added or something. :)

DavidKnight247, 2018-07-09:
I was toying with the thought of doing this at compile-time. It would then generate the correct array without any memory overhead on the device. I’ll take a look at this later on.