OtherCrashOverride/go-play #38: [GB] Video Issue in GameBoy Emulator

puggile, 2018-07-12:
fw version: 20180712
I have video issue in this game:

Wario Land 2:
1) Nintendo logo in menu screen is truncated
2) start new game and see intro, some images are truncated
3) I provide you a save file. 
[Wario Land II (USA, Europe).gb.sav.zip](https://github.com/OtherCrashOverride/go-play/files/2188076/Wario.Land.II.USA.Europe.gb.sav.zip)
Once load, just press "A" + "down" button to see white pixel line on bottom of screen

For your information I tried points 1) & 2) with "Wario Land 2.gbc" on Gameboy Color emulator and I haven't notice issue.
I'm not tried 3) because it's not at the start of the game. If it's necessary I'll tried.

Pokèmon Red Version:
1) Once load Rom see the logo "Game Freak" on intro in center of screen. Is truncated
2) Start new game, then move player left to right or right to left to see video issue. I provide you a save file for easily reproduce issue.
[Pokemon - Red Version (USA, Europe).gb.sav.zip](https://github.com/OtherCrashOverride/go-play/files/2188079/Pokemon.-.Red.Version.USA.Europe.gb.sav.zip)

OtherCrashOverride, 2018-07-12:
Did this occur in previous firmware versions or did it start with the 20180712 release?

puggile, 2018-07-12:
The issue start with fw: 20180708
I didn't have time to write this problems before. Sorry :(

OtherCrashOverride, 2018-07-12:
Resolved by 3e313c61ddb30fde2a6deb56acdf10019c39280a

closed 2018-07-12 by OtherCrashOverride

puggile, 2018-07-22:
I have updated to the latest firmware version ( 20180718-smsplusgx ), but I still have this problems:

Wario Land 2:

I provide you a save file.
Wario Land II (USA, Europe).gb.sav.zip
Once load, just press "A" + "down" button to see white pixel line on bottom of screen

Pokèmon Red Version:

Start new game, then move player left to right or right to left to see video issue. I provide you a save file for easily reproduce issue.
Pokemon - Red Version (USA, Europe).gb.sav.zip