OtherCrashOverride/go-play #39: Suggestion: add quantisation to audio at reduced bit depth

DavidKnight247, 2018-07-13:
I'm uncertain as to how much processing power this would require but it should help to improve sound quality to a degree at reduced bit depth.

Sample C++ code is available [here](http://digitalsoundandmusic.com/5-3-6-simulating-sampling-and-quantization-in-c) though there are probably many other examples available online.

DavidKnight247, 2018-07-13:
Additionally Dither applied to the 16bit signal may improve sound quality at 8 bits.

https://en.wikipedia.org/wiki/Audio_bit_depth

OtherCrashOverride, 2018-07-14:
The quantization is already taking place.  

Since we are dealing with 2 bits of audio at lower volume levels, I do not think dithering would improve it.  Feel free to conduct experiments and report your findings.

DavidKnight247, 2018-07-14:
Ok, I’ll try dithering with the LSB with pseudorandom noise and report back.

dwillmore, 2018-07-19:
What about using the 'remaining error' method?
If you want to output value A, but, when truncated it is B, record the difference A-B and apply it to the next A sample before truncation.  Keep doing that.  It only involves a few simple operations and very little storage.  @DavidKnight247 can you try this?  I don't yet have a build environment set up.