OtherCrashOverride/go-play #40: [GG] Aerial Assault (World) won't load

kamotswind, 2018-07-13:
The emulator seems to execute but it is just a blank screen with no audio. Resetting the emulator with power on while holding start doesn't make a difference.

asiag0, 2018-07-13:
try resetting the system with Menu button instead of Start (Power Switch Off, Hold Menu, Power switch On, Release Menu when Main Menu appears)

this game? [https://imgur.com/a/4G6rKo3](https://imgur.com/a/4G6rKo3)

might just be a bad rom.  the file i have seems to play just fine, though I suck at the game.

Adem1987, 2018-07-13:
No, there are a lot of roms with the [!] tag but they don’t work.

kamotswind, 2018-07-13:
I've reset using both methods, should have mentioned that. It could be a bad rom which I didn't think about since I haven't had that happen before :) I'll give a new one a try and report back.

kamotswind, 2018-07-13:
@asiag0 What firmware version are you on?

OtherCrashOverride, 2018-07-14:
For SMS/GG, the most likely cause of an issue is either timing or lack of mapper support.  I searched but could not locate the mapper used by the game.  If it is a non-standard, support will need to be added.

kamotswind, 2018-07-15:
That's fine. I thought it may be something simple so if it turns out not to be then I'll just close this issue.

OtherCrashOverride, 2018-07-15:
The title works with this:
https://github.com/OtherCrashOverride/Genesis-Plus-GX-crashoverride

So, the answer is somewhere in there.

DavidKnight247, 2018-07-15:
Genesis plus go has a lot of rom mappings

DavidKnight247, 2018-07-15:
Try looking here:
https://github.com/ekeeke/Genesis-Plus-GX/blob/ae28f874021d92f0abd37113a56dcec89b570777/core/cart_hw/sms_cart.c

kamotswind, 2018-07-15:
Looking over that list and doing some more research I downloaded a ROM named: Aerial Assault (Japan) (Rev 1).gg

I never thought to try a ROM that said it was for the Japan region only. However, everything is still in English and it works perfectly. It may be the one labeled "World" has slightly different code that just doesn't work with the current emulator. I think it is safe to say this is a ROM issue and not really something that should be fixed in the emulator right now. I'm going to close this issue but it can always been re-opened if other ROMs are reported to have a similar issue.

OtherCrashOverride, 2018-07-18:
Stay tuned.

OtherCrashOverride, 2018-07-18:
This should now work with the latest release:
https://github.com/OtherCrashOverride/go-play/releases/tag/20180718-smsplusgx

kamotswind, 2018-07-18:
It does, thanks!

closed 2018-07-18 by kamotswind