OtherCrashOverride/go-play #42: How to use firmware.bin from latest releases?

gigawatts, 2018-07-14:
It's not clear how we are supposed to make use of the "firmware.bin" in the last few releases (for example, https://github.com/OtherCrashOverride/go-play/releases/tag/20180712) from this repo.

The firmware update instructions from the Wiki have a bunch of split up firmwares for each partition in the flashall script, so thats pretty clear how to use that.

gigawatts, 2018-07-16:
Got it, that worked :+1:  You might want to make it more clear that the "firmware.bin" file needs to sit on the root of the sd card, not in the "odroid/firmware" directory where I would have expected it to go. It might be helpful to post these steps here in the git wiki, for those (such as myself) that didn't know such forum topics existed.

OtherCrashOverride, 2018-07-18:
There are tentative plans to use "odroid/firmware" for a future feature in the device firmware.

OtherCrashOverride, 2018-07-26:
The "odroid/firmware" folder is now used by the 20180716 device firmware to support multiple applications.

closed 2018-07-26 by OtherCrashOverride