OtherCrashOverride/go-play #43: Remove systems from menu

arelas, 2018-07-21:
Would it be possible to remove systems from the menu if the skeleton does don't exist?

arelas, 2018-07-21:
And also go directly to the games list if only one system exists.

OtherCrashOverride, 2018-07-21:
The solution I am considering is returning to the same spot where the menu was before a title was launched.

hpcodecraft, 2018-07-24:
Returning to the same spot would be a great improvement! Removing unused systems from the menu would be nice too. Maybe it could be hidden when a rom folder is empty? That way users wouldn't have to touch the skeleton folders.

OtherCrashOverride, 2018-07-25:
Currently, I am more inclined to "gray out"/disable a menu than completely remove it.  This would be the case when a ROM folder does not exist or is empty.

If/when future emulators are added, I imagine this issue will be of increasing importance.

ValliantStorme, 2018-07-26:
Moving unused/greyed-out emulators to the end of the list would be good as well

hpcodecraft, 2018-07-26:
If they just were skipped when navigating with the D-pad left/right, that would be good enough for me

OtherCrashOverride, 2018-07-26:
Another option with the latest device firmware would be to package each emulator individually.  This would require some additional code for file selection in the emulators.