OtherCrashOverride/go-play #47: [GB] Video Issue in GameBoy Emulator #38

puggile, 2018-07-23:
Reference to the #38 

I have updated to the latest firmware version ( 20180718-smsplusgx ), but I still have this problems:

Wario Land 2:

I provide you a save file.
[Wario Land II (USA, Europe).gb.sav.zip](https://github.com/OtherCrashOverride/go-play/files/2219340/Wario.Land.II.USA.Europe.gb.sav.zip)
Once load, just press "A" + "down" button to see white pixel line on bottom of screen

Pokèmon Red Version:

Start new game, then move player left to right or right to left to see video issue. Objects in game tremble when the player moves.
I provide you a save file for easily reproduce issue.
[Pokemon - Red Version (USA, Europe).gb.sav.zip](https://github.com/OtherCrashOverride/go-play/files/2219342/Pokemon.-.Red.Version.USA.Europe.gb.sav.zip)

OtherCrashOverride, 2018-07-23:
I isolate the Wario Land 2 issue to the background layer.  It appears to be operating as expected.  Can you check if this issue occurs in other emulators too?  I suspect it is "correct" operation.

puggile, 2018-07-24:
@OtherCrashOverride you are right.
I tried Wario Land 2 with VBA emulator and the white line on the bottom appears in the same point.

I close.

closed 2018-07-24 by puggile