OtherCrashOverride/go-play #53: [Request] Delete game files

ashleyhull-versent, 2018-07-26:
Some games are just bad games. I'd like to delete unwanted games.

rocky1138, 2018-07-28:
What I've been doing is putting my entire roms repository onto the card in a "backup" folder and just copying over the 2 or 3 games I am currently playing to the real roms folder. This works much better also because I no longer have to sift through 45 bazillion games to get to the one I want and the games are all still there if I ever do want to switch them up.

JasonB32, 2018-07-28:
Sift through all the games on a windows emulator then delete them there, then transfer good one's to SD Card. That's what I do.

celentanos, 2018-08-10:
I find the option to delete the games also useful. This can be done, for example, in the menu as an additional point to 'Restart' and 'Play / Resume'. For example, I could not play many games in my childhood, so I do not know which ones are good and which are bad ones. On the PC, I do not want to try this.

dayue1984, 2018-08-27:
I need this too, few basic file manage functions

shar33f, 2018-08-27:
I'd love a really simple file manager that would allow me to simply move roms between folders. For now, I keep entire romsets in a separate folder and then move the ones that I want to play in the correct folder. I carry a small USB-C microsd card reader on a keychain and use that connected to my phone to move roms when needed. Would be great to have a built in file manager right on the odroid go to handle this without having to swap out the microsd card.

OtherCrashOverride, 2018-09-02:
I agree there is a need for file management.  However, I think it should be a stand alone application rather than part of the menu.  This would eliminate the need for other applications to also implement file management.

DavidKnight247, 2018-09-02:
I have found Midnight Commander to be a nice file manager for consoles in
the past. Perhaps this could be ported over.
https://midnight-commander.org

OtherCrashOverride, 2018-09-03:
That was actually my first thought too.  However, not having a keyboard is an obstacle to implementing existing file managers.

shar33f, 2018-09-05:
I was just thinking a very simple UI to move or delete files (not rename them), so no keyboard needed. There is a simple file manager included on my GCW Zero that does basic move / delete using just the dpad and buttons, I can't remember the name but I'll check when I'm back home.

asiag0, 2018-09-05:
@shar33f It's 'Dingux Commander'.  Here's the site for it: http://beyondds.free.fr/index.php?Dingoo-dinguxcommander

Not sure if it's opensource or not.

shar33f, 2018-09-05:
Ah yes @asiag0 that's the one! The source code is on the same link you provided :)

DatOdroidGuy, 2019-11-15:
Could also try a port of FBI from the 3DS since that lacks keyboard yet can delete, move and copy files just from A and B plus the D-Pad