OtherCrashOverride/go-play #57: SMS/GG button mapping

ToddlerPichu, 2018-07-31:
The new SMS/GG emulator switched around the button mapping (was A=1 B=2, now A=2 B=1 ), is there a way to change it back to original mapping?﻿

kamotswind, 2018-07-31:
I just did a quick Google search for the SMS controller and it looks like the left action button on the SMS (2) would best map to the left action button for Nintendo (B) which is why it may have changed.

OtherCrashOverride, 2018-07-31:
The change was made because a game (I don't recall which) depended on the physical location of the buttons (left, right).
[image]

The buttons used can be changed here:
https://github.com/OtherCrashOverride/go-play/blob/04e300aa6ede46c8538761c4a0ac3168305828f9/smsplusgx-go/main/main.c#L658-L659

Since the mapping of B=1 and A=2 match the hardware being emulated, this mapping will continue to be used.

closed 2018-08-10 by OtherCrashOverride