OtherCrashOverride/go-play #59: Game list locks up in new version when many games are present for the system

wvdschel, 2018-07-31:
I have recently upgrade to the following versions:

- [Firmware 20180727](https://github.com/OtherCrashOverride/odroid-go-firmware/releases/tag/20180727)
- [Go Play 20180726-smsplusgx](https://github.com/OtherCrashOverride/go-play/releases/tag/20180726-smsplusgx)

On my SD card I have a couple of hundred Game Boy ROMS, and the system locks up when I try to open the list. It worked fine before.

OtherCrashOverride, 2018-07-31:
This likely due to the change in file sorting.  Previously, the order was numeric: "B" < "a".  Now the order is case insensitive: "B" > "a".

I mentioned on the forum:
> Since the file list is sorted, it is necessary to load all the file names into memory. After this a 'quick sort' is performed. This algorithm, like all fast sorts, is recursive: it takes more memory depending on how 'unsorted' the list is. Deleting and re-copying games changes this 'natural order' and that is why it affects success. Adding, erasing, or renaming a file will also affect success.

Testing is done with 2,724 NES files, 1,424 GBC files, and 1,609 GB files.

OtherCrashOverride, 2018-08-10:
The 20180810 release should address this.

wvdschel, 2018-08-12:
Confirmed fixed :+1:

closed 2018-08-12 by wvdschel