OtherCrashOverride/go-play #62: Sound volume control refinement

DavidKnight247, 2018-08-02:
I don’t like cycling volume in order to mute the device. Usually when you want to mute something you definitely don’t want to make it louder!
1st suggestion. Change to decreasing volume as opposed to increasing the volume.
2nd suggestion. Long press volume button to reduce volume.
3rd suggestion (the one I am currently doing). Put your thumb over the speaker grill until the desired volume is reached!

yaustar, 2018-08-03:
I just go to the menu, change the vol level there and press B to go back to the previous game.

I would actually prefer a button combo to change vol (vol + Up/Down || B/A).

tsndr, 2018-08-11:
In the newest release the even the smallest volume is way to loud for playing at night.

kamotswind, 2018-08-11:
@tsndr what version did you upgrade from?

tsndr, 2018-08-11:
@kamotswind I upgraded from the version mentioned in [that wiki](https://wiki.odroid.com/odroid_go/emulator/firmware_update) `20170702-psram` to `20180726-smsplusgx`

tsndr, 2018-08-11:
@kamotswind I'just updated, to the newest release, but no difference.

kamotswind, 2018-08-11:
@tsndr I think the volume settings changed since 20180702 so that may be what you are experiencing. @OtherCrashOverride may be able to confirm. If you are good with a soldering iron there is a way to reduce the volume levels. https://wiki.odroid.com/odroid_go/silent_volume

Adem1987, 2018-08-12:
> Since the update i have to play on mute all the time because it's way too loud. Can't you just add more volume levels so that it's possible to change the volume more precisely?

Yes. Why had it been changed?

kamotswind, 2018-08-13:
It changed because the previous volume levels caused distorted sound with some games. The volume levels are literally cutting out bits of resolution to reduce the volume. Anything lower then the current lowest setting would be less than 1 bit of audio resolution. Games with a lot of music mixed with sound effects end up having audio issues which was a complaint. Some people want audio quality, and others just want low volume. It is hard to please both with the current hardware limitations.

An external DAC may be an option for you :)  https://www.youtube.com/watch?v=cKPM9KuZo8E

If you have some soldering skills there are ways to reduce the output volume by changing the onboard resistors or putting a resistor in-line with the speaker.

tsndr, 2018-08-21:
So what would be the solution to our problem?

kamotswind, 2018-08-22:
I provided two options in my last reply.

tsndr, 2018-08-22:
An external dac is not practiacal and soldering is not an option. So sadly mute all the time is the only option.

asiag0, 2018-08-22:
Grab an inline volume control from amazon or ebay and just use headphones, sheesh.