OtherCrashOverride/go-play #71: Guide to build the FW file

pmprog, 2018-09-28:
Sorry if this is a bit of a dumb question, but how do you build the .FW file to go on the SD card?

I set up a Lubuntu VMwith the following esp-idf, xtensa-esp32-elf and your odroid go repos.
When I did make in "doom-odroid-go", it created a build directory with a number of files and directories
When I did go-play/odroid-go-common, I got asked a number of questions (that I didn't know what to answer, so I left the defaults?), and again have ended up with a build folder with files and directories

These build directories have two .bin files, an .elf, and a .map; and lots of folders that look like library code/objects; but not a FW file I can put on the SD card.

TIA

explit7, 2018-09-28:
Another question: Are you going to build a Firmware with Doom integrated into the standard Odroid-FW (With the stock Emulators) ?

devsaurus, 2018-09-28:
See https://github.com/23pieces/Odroid-Go-Tools. That built a good firmware file for me.

pmprog, 2018-09-28:
@devsaurus thanks, I'll take a look at that

explit7, 2018-09-28:
I'm not really a dev. and building and compiling is a bit complicated for me.
If someone could send me his ready .fw file - it would be awesome.

1000sun (at) mailbox.org

Thanks

pmprog, 2018-09-29:
@explit7 Why don't you just download it from the official ODroid Go wiki?

explit7, 2018-09-29:
Hi,
I have already the stock odroid firmware from 10.08.2018 downloaded, but i want a fw, where doom and (maybe) other emulators are integrated in the Game Menu

yoursunny, 2018-09-30:
> If someone could send me his ready .fw file - it would be awesome.

They are on “releases” page of each repository, such as 
https://github.com/OtherCrashOverride/doom-odroid-go/releases

OtherCrashOverride, 2018-10-01:
> how do you build the .FW file to go on the SD card?

This is not supported for Go-Play.  For Doom, it is documented on the release page.

There are currently no plans to integrate Doom into Go-Play.  It has not reached a level of maturity where its possible to integrate.  For example, after starting it, you could never return to the menu.  It would need to be playable (including sound and WAD support) before it would be considered.

pmprog, 2018-10-02:
Thanks for the reply. I'll have a play with the doom release and see how I get on.
Cheers

closed 2018-10-02 by pmprog