OtherCrashOverride/go-play #73: Support bluetooth controller(s), keyboard, speakers, perhaps even link cable emulation?

q-killer, 2018-10-01:
It would be great if we could pair a bluetooth controller such as a ps3 gamepad to play two player games simultaneously, a bluetooth keyboard would be very useful also, as would being able to link devices for two player as with the link cable..., ideally this could be done at a low level so both systems are synced on the same game to allow two players simultaneously.

OtherCrashOverride, 2018-10-01:
There are two obstacles preventing this:
1) The ESP32 does not have enough free resources to run a BT stack and emulation at the same time.
2) There is no *host* HID support in esp-idf.  The ESP32 can act as keyboard, gamepad, etc., but it can not currently use a keyboard, gamepad, etc.

q-killer, 2018-10-14:
I've heard directly from espressif systems:

Currently, BT HID profile is on schedule, and it will be launch at the end of this month.

Btw, Our BLE can be as BLE HID Device now.  But can't be BLE HID HOST currently, it is on schedule too, but no launch time.

So if cpu usage can be lowered or is currently optimized enough for certain emulators, perhaps this can be a reality at some point?