OtherCrashOverride/go-play #75: [Request] Game restart confirmation/backs up old save

concreted, 2018-10-13:
Having the ability to restart a game is great, but since it replaces the save state there is the fear of deleting your save by accident. Would be nice if the Restart option backed up your last save, and/or had an extra confirmation screen.

gentleben, 2018-10-13:
Hi,

You might be interested in a fork of go-play I created that contains a simple change to how the menu button functions while a game is being played. A long press (1 second) of the menu button saves the gamestate and takes you back to the menu. A short press just takes you back to the menu without saving the gamestate (i.e. your previous gamestate is still intact).

Here's my go-play fork on Github: https://github.com/gentleben/go-play

explit7, 2018-10-13:
Thanks, will try it

q-killer, 2018-10-14:
Hi gentleben,  thanks for your work on this! 
 Why not integrate this change into the original go-play? Having multiple save games or a confirmation would be great.

gentleben, 2018-10-15:
The original go-play is OtherCrashOverride's official respository. Existing users will be used to the documented menu functionality. If someone would like the changes I have implemented, then they will need to use my fork.

Implementing multiple game saves would be a quite a bit of work - the loading and saving of game states would need to include an additional selection menu of some kind. My C-coding skills are probably not up to the job, and I don't have the spare time to invest (hence the very simple changes I have implemented in my fork). If you know someone who's a good C-coder though, they could create their own fork to implement this.