OtherCrashOverride/go-play #76: [NES] 2 player mode detects 2nd (unusable) controller

q-killer, 2018-10-14:
Second player starts in games as though a second controller is connected, but there's no way to connect a second controller yet, or play as player 2, which would be preferred.

OtherCrashOverride, 2018-10-15:
What NES game is used to reproduce this behavior?

q-killer, 2018-10-15:
I've reproduced this on both smb and smb3

OtherCrashOverride, 2018-10-16:
I believe this is correct behavior.  Someone with a real NES will need to verify that 2 player mode always requires a second controller.

Based on https://wiki.nesdev.com/w/index.php/Standard_controller
It does not appear that NES hardware can detect whether a controller is plugged in or not.

JasonB32, 2018-11-05:
I've owned a real NES ever since inception and it absolutely does not detect or know if you have a second controller plugged in. You can start a two player game with the first controller on any game at all from the best of my knowledge.