OtherCrashOverride/go-play #77: Screen Resolution issue on latest release

dudesky71, 2018-10-15:
Gameboy and Gameboy color do not render at full screen on the October (latest) release. It's 1/4 the size it should be.

gentleben, 2018-10-15:
Hi. While playing a game have you tried holding down "Start" and then pressing right on the d-pad? That should switch scaling on/off.

dudesky71, 2018-10-15:
I have not tried this but i will. I didn't realize this was an option. Where can i find a list of commands/controls for this type of thing? Thank You.

gentleben, 2018-10-15:
This page on the Odroid-Go wiki lists how the various buttons work with go-play:

https://wiki.odroid.com/odroid_go/emulator/usage_go_play

dudesky71, 2018-10-15:
Excellent. I saw that page but didn't scroll down to see all the details. Thank you :)

gentleben, 2018-10-15:
No probs! :-)

kamotswind, 2018-10-17:
@dudesky71 If this is resolved please close the issue :)