OtherCrashOverride/go-play #80: [NES] Please support iNES mapper 30

BenMcLean, 2018-10-26:
[NESMaker](http://austinmckinley.com/8bit/the-tools.html) is a great new tool for creating NES games but most of the games produced (with a few exceptions) are going to be mapper 30. It would be wonderful for them to be playable on the ODROID GO. I think this feature is worthwhile because we're likely to see alot of homebrew on this mapper over the next decade.

Maybe [how RetroArch implemented this](https://forums.libretro.com/t/request-ines-mapper-30-support/9350/9) could be of help? :)

BenMcLean, 2018-10-27:
A good test for mapper 30 support might be the "Troll Burner" game [on this page](http://austinmckinley.com/8bit/the-tools.html)

awakebyjava, 2018-12-25:
I'd like to see support for mapper 30 was well, as I just got the ODROID GO and have been working on my wester NESmaker game for a while. Troll Burner (mapper 2) works find on the Odroid. I know there is a way to add cores, but I don't have the time to dig into it right now. Would be nice to see it supported.