OtherCrashOverride/go-play #81: Some game defects...

phiber1, 2018-10-27:
As I find them:
Nintendo-
Elevator Action: panning is completely broken, screen is doubled in a weird way and unplayable.
Exerion: panning is corrupting the screen on the sides, creating a field of letters.
Stargate: score and weapon status disappears as soon as you move your ship.
Tiger Heli: entire screen flickers badly.

Colecovision-
Pitfall 2: Touching gold bars has no effect, can't get them (sprite priority problem?)

I can report back if I find more. As it stands, very impressive how many games work well! This is my new favorite handheld.

phiber1, 2018-10-28:
Weird thing with Colecovision: I don't know if there's some intermittent state/memory corruption going on, but now Pitfall 2 is totally normal.

Also, I was experiencing a weird sound glitch in Galaxians, also on Colecovision, where the sound would be completely mute UNLESS I shot a missile, during which I could hear all the other sounds as long as the missile was firing. Then mute again. At the moment, Galaxians sound has gone back to normal.

I haven't yet figured out what's causing these glitches, but they seem intermittent.

kamotswind, 2018-10-29:
When I have experienced similar glitches in the past it was caused by a faulty SD card.

phiber1, 2018-10-30:
Nothing that obvious. And the issues I described in the Nintendo games are consistent, independent of the sdcard. Try them yourself.

kamotswind, 2018-10-31:
Sorry I was referring to just the ColecoVision issues. I'll give the NES games a try.

phiber1, 2018-10-31:
I've been able to reproduce this.
If you play with the volume button during game play, the sound messes up in
some colecovision games, like I described with Galaxian. It's consistent.
Cycle through the volume and pick one or two up from mute, and it'll happen.

The other problem I described with Pitfall 2 on colecovision, is with
screen size doubling (start + right). If you double many games, like
Pitfall 2, it looks like the sprite priority completely breaks. You can't
get any gold or other treasures, nor save your friends. You pass right
through them and nothing happens. Games become totally unplayable. Even if
you set the doubling back to normal, during the game it doesn't fix it. You
have to exit with select and return to the game in original small frame,
then sprite priority is normal again. The screen doubling breaks other
games in weird ways as well, since it makes certain collisions undetected.

phiber1, 2019-06-19:
This issue has been open for quite a while. Is no one investigating emulator bugs?