OtherCrashOverride/go-play #91: Repackaging Entire Go-Play firmware

32teeth, 2019-05-06:
Editing core odroid go display (and odroid common) to enable different TFT displays.
Have been able to create FW/BIN for each emulator.

No instructions on how to repackage entire Go-Play FW as single package

pebri86, 2019-05-11:
It's possible to repack everything with different display driver except the launcher, Odroid Go Springboard is closed source, so you can repack but replace springboard with another launcher app from the forum. Go [here](https://wiki.odroid.com/odroid_go/arduino_app) to learn how to use mkfw to pack binary compiled to .fw file.

32teeth, 2019-05-16:
If i do a flash read of the springboard, can i repackage with that?

pebri86, 2019-05-16:
Yes, you can repackage with that, but springboard compiled for odroid go hardware so if you use different TFT display, it won't work correctly because of different driver.

32teeth, 2019-05-16:
> Yes, you can repackage with that, but springboard compiled for odroid go hardware so if you use different TFT display, it won't work correctly because of different driver.

@OtherCrashOverride 
This is the problem i am currently encountering!

It would be nice if we could add different TFT drivers for Springboard

| Description | Results |
| ------------------ | ---------------- |
| 
[image]
 | Launcher (needs display driver) |
| [image] | Game List  (needs display driver) |
| [image] | Game Select (needs display driver) |
| [image]
 | In Game (ALL GOOD) |
| 
[image]
 | ALL GOOD  |

pebri86, 2019-05-17:
We have been talk at the forum for this issue for a while, the springboard is remain as closed source we couldn't do anything other than replace it with open source alternative that someone share on the forum. That's why i use go go launcher as base of my launcher.

closed 2019-10-29 by 32teeth